mhaeuser/Battery-Toolkit #2: Sleep issues

iamfredrik, 2023-09-17:
Hello,

I truly appreciate the effort you put into this project. However, I wanted to bring to your attention some issues I have encountered while using this application.

1. Sleep is not disabled when disabling the Power Adapter

2. Sleep gets disabled when requesting charging to max/full

Expected behavior would be the reverse of the above.

Environment:

MacBook Air M2
MacOS Ventura 13.5.2

Thank you!

mhaeuser, 2023-09-17:
There is a setting for 1., is it enabled? The sleep button in the Apple menu is not greyed out?
2. is expected, charging can’t be regulated when the system is asleep.

iamfredrik, 2023-09-17:
1. Yes, the setting "Prevent your Mac from sleeping when the power adapter is disabled" is checked.

2. Okay, I can understand the behavior when requesting charge to max, but shouldn't it be unregulated when requesting charge to full, i.e. enabling default MacOS settings?

mhaeuser, 2023-10-05:
1. I tried to reproduce it a few times, but it consistently works for me. What happens when you disable the power adapter and then toggle the setting in the GUI once and twice?

2. Well, yes, that could be a sort of micro-optimization, but so far I considered it unnecessary internal complexity. I don't think there is any noticeable impact on power draw by not sleeping. Maybe the drawbacks of disabled sleep are worth avoiding, not sure.

iamfredrik, 2023-10-05:
The sleep button does in fact get greyed out now after a MacOS update and restart. However, in clamshell mode I'm still having the problem that the system goes to sleep despite this and then I have to open the lid to get it to wake up. So I have to use a Keep Awake feature of a third party app to prevent this (or do it manually using terminal). Here's how to correctly disable sleep manually using the terminal so that it doesn't go to sleep in clamshell mode:

# Disable sleep
sudo pmset -a sleep 0
sudo pmset -a hibernatemode 0
sudo pmset -a disablesleep 1

# Enable sleep
sudo pmset -a sleep 1
sudo pmset -a hibernatemode 3
sudo pmset -a disablesleep 0

mhaeuser, 2023-10-05:
Fun. Currently, only the last command is executed. I’ll have to experiment with this some day, but I’m busy with work right now, sorry. :(

mhaeuser, 2023-12-10:
I dealt a bit with disabled sleep as part of issue #3 and I cannot reproduce the issue, the system stays awake perfectly fine. *Sometimes* it goes to sleep immediately when sleep is re-enabled (as part of enabling the power adapter), but that is a macOS bug that has been discussed online various times (and does not affect Intel Macs, for example). As for your commands:

> sudo pmset -a sleep 0

According to the documentation I read, this should disable automatic sleeping based on a timer. I do not think it should be relevant here.
> sudo pmset -a hibernatemode 3

This is more of an instruction on what to do when sleeping rather than specifying whether to sleep.
> sudo pmset -a disablesleep 0

This is currently executed (well, not really because the tool uses the IOPM APIs directly, but it should replicate the behavior).

Of course, the first two might somehow influence the bug I mentioned, if that is what you hit. Can you please explain what the additional commands are supposed to achieve precisely and maybe give reproducible steps to compare their behavior?

iamfredrik, 2023-12-10:
The command "sudo pmset -a sleep 0" on a Mac sets the system sleep timer to 0 minutes, effectively disabling automatic sleep.

The command "sudo pmset -a hibernatemode 0" on a Mac sets the hibernation mode to 0, which disables hibernation. The default value is 3 which means the system will store the contents of memory to disk and then go into a low-power mode.

The command "sudo pmset -a disablesleep 1" on a Mac enables the option to disable sleep. This means that sleep is now disabled.

All the above commands are required when using Battery-toolkit in clamshell mode using a Macbook Air M2. Otherwise the system will go to sleep and it won't wake unless opening the lid, which is very annoying when using a vertical stand.

mhaeuser, 2023-12-10:
Well, you basically repeated what I described. I do not understand why the hibernation mode matters at all in this scenario. Neither do I understand why automatic sleep with a timer would trigger when sleep is globally disabled. Also I need steps to reproduce the exact issue (do you wait some time till the sleep timer triggers?), because just disabling the power adapter keeps the machine awake with the lid closed.

iamfredrik, 2023-12-10:
Those commands have been suggested all over the internet to keep Macs awake. There are utility apps like [Amphetamine](https://apps.apple.com/au/app/amphetamine/id937984704?mt=12) that does the same thing. Macbooks go to sleep in clamshell mode even when sleep is disabled, because they are not designed to operate in clamshell mode using battery power.

Steps to reproduce.

1. Put Macbook in clamshell mode
2. Disable the Power Adapter using Battery-toolkit
3. Do nothing and wait until external monitor goes to sleep
4. Try to wake it up using external keyboard and mouse (at least in my case it won't wake unless opening the lid)

mhaeuser, 2023-12-10:
Well, in my years as developer for the Hackintosh community, I've seen tons of *awful* advice and wrong explanations all over the internet. One example is what is also posted all over the internet to re-enable sleep, namely setting "sleep" back to 1 - as if it was a boolean value, while (as we both seem to agree) it is indeed a timeout value. So to re-enable sleep, the suggestion is to set it to a 1 minute timeout, rather than providing a default value (20) or explaining what value to choose.

Anyway, I finally reproduced the issue. I'm afraid this is an extension of the Apple bug I mentioned previously. Namely, once the display(!) goes to sleep, it indeed does not wake without opening the lid. And when you open the lid and then close it again, it appears to cache the usual sleep request. When you re-enable the power adapter afterwards (and thus re-enable sleep), the system will immediately sleep, even without any acute reason for doing so. I confirmed the system indeed does *not* sleep when triggering the display sleep issue by playing music till reaching display sleep. The keyboard remained lit, the music kept playing, but the system would not wake the external monitor till opening the lid.

While I understand this is annoying for you, I will not address this as part of this tool. The only concern with toggling the power adapter is system-wide sleep to enable usability in clamshell mode when the power adapter is disabled. I do not want to introduce workarounds with unexpected behavior (disabling display sleep) to work around Apple bugs that may be fixed in the future.

closed 2023-12-10 by mhaeuser

iamfredrik, 2023-12-11:
Fair enough. The best bet for now is to use the [Amphetamine](https://apps.apple.com/au/app/amphetamine/id937984704?mt=12) app and set a custom trigger to disable sleep when the power adapter is disabled.