mhaeuser/Battery-Toolkit #7: Battery Toolkit not working properly

SohamJa1n, 2024-03-12:
[image]
When I try to start it, it gives me this error, even after restarting my mac.

I'm on Sonoma 14.4

mhaeuser, 2024-03-12:
Is the background task enabled in System Settings > General > Login Items > Allow in the Background > Battery Toolkit? What Mac Model is it?

SohamJa1n, 2024-03-12:
Yes it is enabled and I'm on an M2 Air

mhaeuser, 2024-03-12:
Weird, I've tested it myself on M1 Pro and M3 Pro, and various other models are tested by others. I'm afraid you'd have to perform some debugging yourself (e.g. checking Console > me.mhaeuser.batterytoolkitd when the background daemon is launched, etc.) - I unfortunately don't have the time to write better debugging stuff at the moment.

SohamJa1n, 2024-03-12:
Alright no worries, it's odd because other apps such as BatFi don't work either so I'm thinking its a localised issue.

Will need to test later on.

mhaeuser, 2024-03-12:
Now that's bizarre. It's not impossible there is some unexpected hardware configuration going on. The daemon quits when it finds an unsupported configuration, but it should display a different GUI error in that case. Either the daemon is borked by some system configuration issue, or you are hitting an odd edge-case where it quits in a different error path than incompatibility. Check Console, really.

mhaeuser, 2024-03-12:
No, that is just noise, I think. You have to log when the daemon is started (ideally, use the uninstall script and make sure the daemon is terminated with Activity Monitor, then start Console logging, and then start Battery Toolkit to monitor the logs).

SohamJa1n, 2024-03-12:
whats the uninstall script?

mhaeuser, 2024-03-12:
https://github.com/mhaeuser/Battery-Toolkit/blob/main/uninstall.sh

SohamJa1n, 2024-03-13:
What should I be looking for in the logs?

mhaeuser, 2024-04-04:
Someone else encountered this and the issue was a stray BTM record that prevented the latest version to update the daemon. Battery Toolkit should be completely uninstalled as follows:
1. Quit the app fully.
2. Run uninstall.sh.
3. Delete the app completely (and empty Trash).
4. Close System Settings, if it is open, and start it anew. Go to General > Login Items and make sure Battery Toolkit is NOT present.
5. Reboot.
6. Re-download and re-try.

If at step 4 Battery Toolkit is still present, you have a stray record as well. You can confirm this by running this command:
`sudo sfltool dumpbtm | grep batterytoolkit`

To get rid of the record, there unfortunately is no straightforward command. You need to locate the supposed location of the stray record with:
`/System/Library/Frameworks/CoreServices.framework/Frameworks/LaunchServices.framework/Versions/A/Support/lsregister -dump | grep "Battery Toolkit"`

If Battery Toolkit exists at the printed location, delete it. If it does not, copy it there and reconstruct the exact structure. Use `ls` to make sure the location is an exact match. Then, delete it again. Deleting an application usually invokes macOS’ garbage collection for such records, which is what we need to purge it. Finally, resume at step 4.

If you still encounter issues after that, as I said, post a proper log file.

closed 2024-04-04 by mhaeuser

SohamJa1n, 2024-04-16:
This worked thanks