mhaeuser/Battery-Toolkit #8: Context surrounding "Prevent your Mac from sleeping when the power adapter is disabled" boolean

peterklingelhofer, 2024-03-17:
First of all, thank you so much for offering this FOSS equivalent of Al Dente, this is significantly simpler and perfect for my purposes (I have a monitor which doesn't allow me to disable charging when plugged in - baffling why they didn't add that option to disable it - so this is huge).

[image]

I saw this option, and it wasn't immediately clear why/how this might be useful, or why it was defaulted to true. What is the typical use-case/justification for this option? I'm happy to submit a PR to add a mouseover tooltip to better explain this, but I just didn't quite understand it myself enough to know what to add. It would be as simple as adding `.help("<Tooltip text>")`.

mhaeuser, 2024-03-17:
Right, I guess it’s not really intuitive. Unlike previous Macs, which allowed to hardcode a maximum threshold, late Intel and AS Macs support only an on/off toggle for charging. To guarantee the maximum threshold is respected, the tool must monitor the battery level in software. This is not possible during sleep, hence the option.

I’m not sure a tooltip is the best solution, but I don’t have the time right now to think about it much.

peterklingelhofer, 2024-03-17:
Ah, I can see things being tricky with clamshell mode. Put up a quick change to offer this context in tooltip form, but feel free to disregard/ignore/close it if this isn't the best way to address it (if you don't think it needs to be addressed that's totally understandable too): https://github.com/mhaeuser/Battery-Toolkit/pull/9

closed 2024-05-11 by mhaeuser

rduttshukla, 2024-11-12:
Will the system sleep when not in clamshell mode? Can we somehow detect if the system is in clamshell mode and prevent sleep only in that case? Apart from toggling the switch manually for these scenarios, I believe auto detection can be a good option

mhaeuser, 2024-11-12:
But why would you disable the power adapter outside of Clamshell mode? I’d need to see whether there is a way to detect this, but I’m not aware of any so far.

rduttshukla, 2024-11-28:
It would be useful in overnight charging scenarios with lid closed. I use my laptop during the day and charge it overnight. I don't want my laptop to stay awake with lid closed after charging till 80% to avoid unintentional battery drainage.

Although I've seen that the battery drainage doesn't happen unless the laptop is on heavy load. I use a slow charger to charge my laptop during the night by the way (~10W charger).

rduttshukla, 2024-11-28:
I believe I misunderstood what the option is about.

"Disable charging" vs "Disable Power Adapter" are two different modes, got it.