mhaeuser/Battery-Toolkit #10: Battery Status

laltm, 2024-04-01:
It would be nice to implement the battery charge percentage into the status bar icon.

Maybe like Apple did it on iOS. That way, you could throw out the default icon and save some space in the status bar. 
I have no Idea how to do this but I think it would be super neat :)

rogueda, 2024-07-22:
@laltm I use Bartender to hide the Battery Toolkit icon as once i have it setup it runs really well in the background, then use iStat to display a more useful battery icon.
iStat has been involved in some controversy lately, so you may want to use one of the alternatives.

mhaeuser, 2024-07-22:
@rogueda You don’t need to use Bartender, you can simply quit the GUI. All actual logic is in the background daemon.

laltm, 2024-07-23:
> some controversy lately
@rogueda 
Hi, thanks for the suggestions!
I actually have iStat, but didn't like the way their icon looks. Can you tell me what the controversy was about?

trevyn, 2024-09-24:
I think Bartender was the one with the controversy.

franzf, 2024-10-20:
> @rogueda You don’t need to use Bartender, you can simply quit the GUI. All actual logic is in the background daemon.

Would you be able to add explanation to the GUI somewhere? I was going to leave this running on the menu bar at all times before checking this issue.

mhaeuser, 2024-10-25:
@franzf This also implies the opposite issue, where people quit the GUI expecting the background process to quit as well. Would a prompt with an explanation that the background activity remains active when quitting resolve your suggestion? You have to try quitting to see it, so it's not perfect.

alexisgaziello, 2025-01-10:
> @franzf This also implies the opposite issue, where people quit the GUI expecting the background process to quit as well. Would a prompt with an explanation that the background activity remains active when quitting resolve your suggestion? You have to try quitting to see it, so it's not perfect.

I think that would be an cool addition. I wasn't aware of it. And maybe even a toggle to keep the daemon on/off when quitting the app.

mhaeuser, 2025-01-10:
Such a toggle would be dirty, because the GUI is a user process and the daemon is a root process (shared across all users). What I could and might do is adding a “Pause Background Activity” command to the menu, but so far nobody really asked for it or provided a use-case.

alexisgaziello, 2025-01-10:
Can you turn off the daemon without uninstalling?

mhaeuser, 2025-01-10:
From System Settings > Login Items, yes.