mhaeuser/Battery-Toolkit #12: Homebrew

Abdullah16M, 2024-05-01:
It would make installing and uninstall the program much simpler and more accessible

Abdullah16M, 2024-05-02:
Also it will provide a convenient way for updates!

For more info: https://docs.brew.sh/Adding-Software-to-Homebrew

mhaeuser, 2024-05-11:
Thanks! Blocked by: https://github.com/orgs/Homebrew/discussions/5357

rcambrj, 2024-06-10:
Thanks for linking the blocking discussion.

> The only way seems to be to indeed bundle a script that would only execute this one command, but that would feel a bit awkward

Would you consider doing that for now, until that discussion is resolved?

simono, 2024-10-03:
@mhaeuser Any news on this topic? I'ld like to install Battery-Toolkit via Brew. Thanks 😄

mhaeuser, 2024-10-03:
No - nobody responded to the blocker and I haven’t gotten around to see whether there is an alternative way.

nhymxu, 2024-10-09:
Don't have time to try again with homebrew casks repo (sometimes they reject)

So I create homebrew taps for myself.
Anyone want to install via brew can try: https://github.com/nhymxu/homebrew-tap

Note: manual update when have new version right now (because not have time to write auto-update function)
So if @mhaeuser release a new version, feel free to ping me or create PR on my repo.

Or wait @mhaeuser create other tap and handle update when release

johnfernow, 2024-12-08:
@nhymxu if I'm not mistaken, the [blocking issue described here](https://github.com/orgs/Homebrew/discussions/5357) applies to your cask as well, right?  

If that's the case, if we want to uninstall, should we just download and run [uninstall.sh](https://github.com/mhaeuser/Battery-Toolkit/blob/main/uninstall.sh) instead of `brew uninstall nhymxu/tap/battery-toolkit`?  

If so, that's fine by me, I don't anticipate needing to uninstall unless a future MacOS update breaks _Battery Toolkit_, but I just wanted to know and make sure.

mhaeuser, 2024-12-08:
@johnfernow Yes, it does. You can also use the GUI “Disable Background Activity” option, but uninstall.sh works too.

mhaeuser, 2024-12-11:
@clns In the menu bar > Battery Toolkit. It might feel a bit hidden, but it’s not a “quick command” like the rest. I should expand the README some day…

mhaeuser, 2025-03-06:
Notarization appears to be mandatory for ARM64 Homebrew casks, so this will be blocked indefinitely:
```
audit for battery-toolkit: failed
 - Signature verification failed:
/private/tmp/cask-audit20250306-54865-2f5dve/Battery Toolkit.app: rejected

macOS on ARM requires software to be signed.
Please contact the upstream developer to let them know they should sign and notarize their software.
battery-toolkit
  * line 7, col 2: Signature verification failed:
    /private/tmp/cask-audit20250306-54865-2f5dve/Battery Toolkit.app: rejected
    
    macOS on ARM requires software to be signed.
    Please contact the upstream developer to let them know they should sign and notarize their software.
```

wangl-cc, 2025-03-06:
If adding it to homebrew-cask is impossible, is it possible to have an ‘official’ tap? A cask doesn’t need to provide binary bottles, so hosting a tap should be enough for easy installation and uninstallation.

mhaeuser, 2025-03-06:
@wangl-cc I didn't get to thoroughly test it, but this might work: https://github.com/mhaeuser/homebrew-mhaeuser

wangl-cc, 2025-03-06:
> I didn't get to thoroughly test it, but this might work: [mhaeuser/homebrew-mhaeuser](https://github.com/mhaeuser/homebrew-mhaeuser?rgh-link-date=2025-03-06T22%3A51%3A07.000Z)

Thank you for your efforts. It works for me.

mhaeuser, 2025-03-15:
Official Homebrew remains blocked indefinitely. Meanwhile, the private tap seems to work, so closing.

closed 2025-03-15 by mhaeuser