mhaeuser/Battery-Toolkit #14: Option to hide menu bar icon

samcarter, 2024-05-15:
Thanks a lot for this handy tool!

It is one of these tools one sets up once and then can basically forget about it because it just keeps doing its job in the background (besides updates every now and then). 

It would be nice to have an option to hide the menu bar icon. This would safe space in the menu bar for users on small displays. 

Some other applications like https://github.com/objective-see/LuLu or https://github.com/rxhanson/Rectangle offer such an option. To still be able to interact with the application, they will open the setting window if the application is started while it is already running.

mhaeuser, 2024-05-15:
The actual logic is handled by a daemon. You can quit the GUI application and it will continue working.

closed 2024-05-15 by mhaeuser

samcarter, 2024-05-15:
@mhaeuser Thanks for your quick reply! This makes getting rid of the menu bar icon much easier than I thought :)