mhaeuser/Battery-Toolkit #15: Intel CPU support?

davidkaufman, 2024-05-17:
What would be needed to support Intel-based macs too?

I use my mid-2018 Intel Macbook Pro as a desktop computer 100% of the time and have had to replace the battery TWICE now, because over time the battery swells up dangerously when it's plugged into power 24/7/365.

So I'd love to help work on bringing this very cool and much-needed app to all the users of older Intel-based Macbook Pro users too!

Thanks!

mhaeuser, 2024-05-17:
Intel machines use different SMC keys and sometimes even logic (charging on/off vs explicit threading on SMC-level). I don’t have an Intel machine, so I can’t test support and am not interested in implementing it myself. If you want to PR it, AlDente Free has some documentation of how it works on Intel.

rodcord, 2025-06-18:
Try this one: 

https://github.com/js4jiang5/BatteryOptimizer_for_Mac?tab=readme-ov-file