mhaeuser/Battery-Toolkit #19: Settings still active after deleting the app.

Arstidir, 2024-05-23:
I had this app installed for a while but decided to delete it. Unfortunately the settings I made (Charge until 85%) a still present. 
I restartet my Mac several times. 

How do I get the default Apple settings back?

mhaeuser, 2024-05-23:
Locate batterytoolkitd, where is it if you deleted the app?

Arstidir, 2024-05-23:
How should I locate it if I deleted it? 
Where? I draged it in the trash and emptied the trash

mhaeuser, 2024-05-23:
Activity Monitor, top/ps, launchd, … Deleted binaries do not execute, so it is not deleted. :)

Arstidir, 2024-05-23:
I don't know what else I could say. It's not active in the activity monitor. And I deleted your app like you would any app on macOS. If your app requires different methods to delete it you should mention it.

mhaeuser, 2024-05-23:
It doesn’t generally require that. On older versions of macOS, you have to disable Background Activity first, but you would see the daemon in Activity Monitor then. On Ventura and up, this is not necessary. If it is not running, then the app is not at fault, any changes are volatile and reset when the machine is rebooted.

Dealbergaria, 2024-05-30:
Hi, I moved the Battery Toolkit app to the trash, the followed the instructions in uninstall.sh and executed those comments in Terminal. For the first two commands, sudo rm /Library/LaunchDaemons/me.mhaeuser.batterytoolkitd.plist and sudo rm /Library/PrivileqedHelperTools/me.mhaeuser.batterytoolkitd, Terminal responded with "No such file or directory exists", all the others seem to have run successfully. I restarted the Mac, but it still retains the Battery Toolkit settings (charges up to 80% max). How can I completely remove it?

mhaeuser, 2024-05-31:
Same responses I gave above, this is literally impossible. Check whether the daemon is running and where it is located, otherwise it’s not Battery Toolkit’s doing.

Dealbergaria, 2024-05-31:
I fixed it. Even after running the terminal commands in the uninstall.sh, there were still some files left which I found after searching for mhaeuser, such as pilists, me.mhaeuser.battervtoolkitd, me.mhaeuser.batterytoolkitd.dSYM, and me.mhaeuser.batterytoolkitd.yml. After deleting them it all worked ok, thank you.

mhaeuser, 2024-05-31:
@Dealbergaria If you used sudo to invoke the script (it uses sudo internally where appropriate), you may indeed have a left-over user configuration plist. dSYM are debug symbols that come with the debug release, so where-ever they were, you extracted them there from the tarball. It's odd you found me.mhaeuser.batterytoolkitd, where was it? Pre-Ventura, it should be in the location the script didn't find. For Ventura+, it should be in the app bundle you said you deleted. yml files are not used at all, not sure what that is about.

Dealbergaria, 2024-05-31:
Unfortunately I didn't take note of their location, this is the only screenshot I have from yesterday
[image]

Dealbergaria, 2024-05-31:
BTW, if I want to re-install it, should I download the Battery-Toolkit-1.4-dSYMs.zip or the Battery-Toolkit-1.4.zip version?

mhaeuser, 2024-05-31:
Non-dSYM, dSYM is for debugging.

Dealbergaria, 2024-05-31:
thank you, that was the one I had originally installed, maybe that was the source of the problem ;)

closed 2024-06-14 by mhaeuser

G2Jose, 2024-08-08:
I actually had the same issue after running `uninstall.sh`. I noticed that my battery wouldn't charge beyond 80%, and it would drain all the way to 70% before charging again (the very same settings I used with BatteryToolkit). 

I really commend this tool and what it aims to do, but I'd highly recommend these changes: 
1. Fixing the uninstaller so it removes _all_ traces of the application and settings
2. Adding instructions on how to run this for less technical folks

mhaeuser, 2024-08-08:
@G2Jose I am well aware of these shortcomings, but I cannot address them properly.

1. cannot fully be fixed simply because Apple doesn't have sane APIs to remove Battery Toolkit from BTM (this also caused other issues where after uninstalling the tool failed to re-install). As for "leftovers" issues, I cannot make a clear call, because I did not get any details. For Ventura+, no files are installed anywhere (due to the SMAppService API). So if there are leftovers, they are in the app bundle you put somewhere. The script cannot know where you put (copies of) the app.

2. is more or less a consequence of 1. and the shaky nature - after all, the tool is not notarized. I would prefer to keep things a bit "non-trivial" till all rough edges are smoothened. Then I'd also distribute BT via homebrew, which also would enable automatic updates.

G2Jose, 2024-08-08:
I appreciate the context @mhaeuser. 

For (1): 
- I'm on Sonoma
- I didn't move copies of the app anywhere (I simply downloaded the release, opened the zip, moved it to Applications)
- Unfortunately I don't have a complete list of all the remnants I found, but in case it helps at all, looking through my shell history here are a few I found: 

```shell
/private/var/folders/8c/g3_5h3js78x1f_l2x9kkcn_r0000gn/C/me.mhaeuser.BatteryToolkit
/Applications/Battery Toolkit.app/Contents/Library/LaunchDaemons/me.mhaeuser.batterytoolkitd.plist
/Applications/Battery Toolkit.app/Contents/Library/LaunchDaemons/me.mhaeuser.batterytoolkitd.plist
/Applications/Battery Toolkit.app/Contents/Library/LaunchServices/me.mhaeuser.batterytoolkitd
$HOME/Library/Preferences/me.mhaeuser.BatteryToolkit.plist
/private/var/folders/8c/g3_5h3js78x1f_l2x9kkcn_r0000gn/C/me.mhaeuser.BatteryToolkit
/private/var/root/Library/Preferences/me.mhaeuser.batterytoolkitd.plist
/System/Volumes/Data/Applications/Battery Toolkit.app/Contents/Library/LaunchDaemons/me.mhaeuser.batterytoolkitd.plist
/System/Volumes/Data/Applications/Battery Toolkit.app/Contents/Library/LaunchServices/me.mhaeuser.batterytoolkitd
/System/Volumes/Data/Users/$USERNAME/Library/Preferences/me.mhaeuser.BatteryToolkit.plist
/System/Volumes/Data/private/var/folders/8c/g3_5h3js78x1f_l2x9kkcn_r0000gn/C/me.mhaeuser.BatteryToolkit
/System/Volumes/Data/private/var/root/Library/Preferences/me.mhaeuser.batterytoolkitd.plist

```

G2Jose, 2024-08-08:
Making a PR with some instructions RE: uninstalls here - https://github.com/mhaeuser/Battery-Toolkit/pull/31

mhaeuser, 2024-08-09:
I commented the PR here: https://github.com/mhaeuser/Battery-Toolkit/pull/31#issuecomment-2277379703

Your file path dump is part of the reason I’m hesitant about uninstall issue reports and documentation.

Had you deleted the application (which you should do in all cases) and rebooted, there would have been no way for limiting to still be active. As per my comment on the PR, you would have a stray AuthServices right and the settings would still be in place, but the system would behave as if the application was fully uninstalled just from that. Further, had you run uninstall.sh (as you said you did), the right and the preferences would also be purged. That’s pretty much all files from your history. The only exception is the /private/var/folders thing, which I don’t know exactly what it is, but appears to be some sort of cache? This is not created or maintained by Battery Toolkit itself and also should cause no effects.

Considering this, I’m not sure what the issue to fix is exactly. Did you only run uninstall.sh and not delete the application? Even that should do the trick to disable background activity.