mhaeuser/Battery-Toolkit #21: Background Activity Issues

shubh30112003, 2024-06-15:
[image]
[image]

I tried uninstalling and installing again but it didnt work.

mhaeuser, 2024-06-15:
macOS 15 is a buggy beta, I’m not sure what you expect. I heard from another user who hit the same issue and a simple reboot fixed the issue. In case the system is now tainted, try these steps: https://github.com/mhaeuser/Battery-Toolkit/issues/7#issuecomment-2036664963

closed 2024-06-15 by mhaeuser

shubh30112003, 2024-06-15:
@mhaeuser Thanks for prompt reply.  I did delete with every file but didnt work, I tried again. no success

mhaeuser, 2024-06-15:
The steps entail more than simply deleting files, and more importantly, they entail two different routes based on an unobvious condition I have no control over. Please read them thoroughly.

shubh30112003, 2024-06-15:
i meant i did run command. Man i loved the app. until 15 becomes stable i cannot use the app

crystall1nedev, 2024-06-24:
Sequoia beta 2 seems to fix the issue for me. This exact situation is why it's a good idea not to run beta software if you aren't ready to deal with software breakage.