mhaeuser/Battery-Toolkit #22: Battery discharges with power adapter enabled

Azirine, 2024-06-15:
Under a ~30W load with a 96W charger, the battery discharges to the lower limit set in the app then charges up again. Power adapter is set to be enabled. Expected behaviour is all power will be taken from the charger without draining the battery.
[image]

mhaeuser, 2024-06-15:
Well, are you sure it’s a consistent 30 W load? Macs, especially the 16”, will use the battery if AC is not enough. Battery Toolkit has no control whatsoever over this behaviour.

Azirine, 2024-06-15:
I have a 14" MBP which should not be able to use more power than the 96W charger can provide. ~30W is seen under Mx Power Gadget, which along with monitor, etc. (monitor is set to medium-low brightness) should only be around ~40W.

mhaeuser, 2024-06-15:
I can’t find any hard facts on this right now, but I think it can very well use more than 96W, especially the Max models. The 16” just has a better thermal headroom to sustain the high clock speeds (and supports faster charging), I’m not aware of it having a larger maximum theoretical power draw. And again, the tool cannot influence this behaviour either way.

closed 2024-06-15 by mhaeuser

Azirine, 2024-06-15:
To show I was not using more power than the adapter, I let the battery charge under the same load, and despite being under load the battery was still charging with 13 W.
[image]