mhaeuser/Battery-Toolkit #32: Do not depend on local Administrator rights to trigger features and update charging thresholds

Kenimore, 2024-08-28:
Currently every action (Disable Power Adapter, Charge to Full and Charge Maximum) in Battery Toolkit dropdown menu prompts for local Administrator rights.
Changing Power settings, i.e. adjusting lower and maximum limits - acts the same, triggering Administrator rights prompt.

I wonder if decoupling daemon state and configuration storage into more easily accessible user settings location (for example, ~/Library/...) will do the trick?

mhaeuser, 2024-08-28:
This is not a technical limitation - quite the opposite, the prompts are entirely internal privilege boundaries unrelated to permissions to system resources. A location like ``~/Library`` cannot work simply because it doesn't make sense to configure an overall system power state per user. The prompts are in place to guard against unauthorized users making changes to how the system behaves (otherwise, even the Guest user could launch the Battery Toolkit GUI and make system-wide changes). However, the current system is a bit of a crutch. I do not like the technical implementation at all and requiring a password prompt when charging to maximum or alike is probably overkill. I wanted to look into reworking this some day and will definitely consider making it less verbose. However, for making persistent changes like changing the thresholds, I think the password prompt is very well warranted.

thomasfire, 2024-09-26:
I think this can be a configurable option. For example those folks who do not have other people in access to their mac definitely would be happy to disable the admin prompts.

mhaeuser, 2024-10-27:
Addressed by https://github.com/mhaeuser/Battery-Toolkit/commit/f157b3e6dd77965139b17dda2430373ace4df557

closed 2024-10-27 by mhaeuser