mhaeuser/Battery-Toolkit #33: Disabling Power adapter puts computer to sleep in clamshell mode

slvrscoobie, 2024-08-30:
Ive done this before on previous versions, but on MacOS 15.1 Beta 3 at least I tried to turn off the power adapter and my monitors blanked out. Im using a powered docking station with the MBP in clamshell mode. 
I tried to unplug and replug but no luck. If I open the screen it works, but as soon as I close it, the monitors blank out. Turning the power adapter on again, and it works as expected.

mhaeuser, 2024-08-30:
Is “Prevent for Mac from sleeping when the power adapter is disabled” enabled or disabled?

slvrscoobie, 2024-08-30:
It is enables.

mhaeuser, 2024-08-30:
Bizarre. I've just tried on macOS 15 Beta 8 and it works fine. Can you check whether sleep is actually disabled, i.e. is the "Sleep" option in the Apple menu greyed out or enabled?

slvrscoobie, 2024-09-01:
welp. I guess you can close this, I must have updated it or something as disabling before worked as expected, but it didnt yesterday. Today, after confirming 'prevent Mac from sleeping' was enabled, it's working as expected.

closed 2024-09-01 by slvrscoobie