mhaeuser/Battery-Toolkit #39: Request - Remove dock icon

M3MaxBruh, 2024-09-24:
Xcode dummy here. This would be great if it didn't open a dock icon when it launches. I'm no good with xcode, but it seems straightforward enough to accomplish, if you know how to navigate xcode, which I don't. I keep getting 'Build Failed' and I'm not sure why...

Anyway, here's what I think it needs:

info.plist needs another key set to Application is agent (UIElement) with the value set to yes.

From what I've gathered, this should make it run as a background agent and launch with only the menu bar icon, instead of launching a dock icon. 

Not too big of a deal, but would be a great UI improvement not having hide the dock icon every time it launches.

mhaeuser, 2024-09-24:
This is intentional. Menu bar extras are not guaranteed to be displayed, so Battery Toolkit can be a focussed application without windows, so that you can trigger commands and access settings from the menu bar. As soon as it loses focus, the icon is automatically hidden.

closed 2024-09-24 by mhaeuser

M3MaxBruh, 2024-09-24:
Ahh, ok. I suppose I didn't play with it enough to see that in action. Thanks for the quick response though :-)