mhaeuser/Battery-Toolkit #41: Not opening

Syiana, 2024-10-07:
Hello, 

I try to bring Battery-Toolkit to work but when I place it in Applications and try to open it, I see the icon for 5 seconds on my dock and then it disappears.

I use the latest Mac os 15.0.1

mhaeuser, 2024-10-07:
And there is no Menu Bar Extra and you don’t see it active in the Menu Bar? There is no window opening by default as it’s by design a background app.

Syiana, 2024-10-08:
I was initially confused because I expected a GUI to appear when opening the app from the Applications folder. However, after logging into my Mac today, I noticed the battery icon in the Menu Bar! Thanks :)

However, it would be great to have an option to hide the icon and open the app directly from the Applications folder.

closed 2024-10-08 by Syiana