mhaeuser/Battery-Toolkit #42: Setting icons are being stretch on external display

alvin96, 2024-10-08:
I just found a UI glitch on my company's external monitor. The icons in the settings menu are stretched. Nonetheless, thank you for this awesome open-source software!

[image]

This is the display model and the resolution I set:
[image]

mhaeuser, 2024-10-08:
Do any other applications have this issue? These are native buttons with native icons - super weird.

alvin96, 2024-10-08:
> Do any other applications have this issue? These are native buttons with native icons - super weird.
Seems like one of my application is also experiencing same issue, likely related to macOS Sequoia. There are no issues on macOS Ventura, so it could be something specific to Sequoia’s display handling. Hopefully, it gets resolved in a future update! Thanks for your fast response. Love this app
[image]

closed 2024-10-08 by alvin96

mhaeuser, 2024-10-08:
Yes, Crapple trashOS Bugquoia… I moved to 15.1 Beta due to the bugs with System Extensions and network filters, but only the former was resolved (until yesterday)… Sigh.

alvin96, 2024-10-08:
> Yes, Crapple trashOS Bugquoia… I moved to 15.1 Beta due to the bugs with System Extensions and network filters, but only the former was resolved (until yesterday)… Sigh.

Sigh... I completely agree, recent OS updates seem to prioritize new features over stability, which can be frustrating. Innovation should always build on a solid foundation of stability without sacrificing reliability. It’s hard when progress feels like a tradeoff!