mhaeuser/Battery-Toolkit #44: Suggestion: Change confusing language

0Commandr, 2024-10-10:
"Charge to full" and "charge to maximum" are too vague when used in combination. I would suggest changing the language to something like "Charge to 100%" and "Charge to limit". I often forget which to use (full vs maximum) because I don't interact with the app that often. 

(sorry if this suggestion is better suited somewhere else, not a frequent github user)

mhaeuser, 2024-10-10:
That’s already staged. I called it “Charge to Threshold”, but maybe “Limit” is even better. I’m awful with intuitive names.

ECBicalho, 2025-02-22:
Instead of terms, it could display the desired percentage:
Charge to 100% now
Charge to <userChoice%> now

mhaeuser, 2025-02-22:
Hmmm… I will not use “100 %” to match with Apple’s terminology, but for the threshold this sounds intriguing. I’ll think on it - thanks!

mhaeuser, 2025-03-05:
Addressed by https://github.com/mhaeuser/Battery-Toolkit/commit/f93677e2c8e0186ab71506e9af7bb3125d36e759

closed 2025-03-05 by mhaeuser