mhaeuser/Battery-Toolkit #46: A few questions about BT functioning

Sangeppato, 2024-10-22:
First of all, thank you for this amazing application.

I have a few questions regarding its functioning, just for the sake of my curiosity (I hope not to annoy you with them).

It is my understanding that:
• Every time the battery charge percentage changes, an event is generated and BT will check whether it has to enable/disable charging according to it.
• If the laptop is plugged in, this also works then the laptop's lid is closed and there are no external displays connected, since the computer is not actually sleeping as long as it's connected to power (is this true?).
• Such event monitoring is paused when the laptop is running on battery, and the only monitoring that occurs in this case is whether the power adapter gets connected (so the power usage of BT should be minimal).

Is all this correct?
Thank you again

mhaeuser, 2024-10-22:
1. Yes.
2. Not quite. Sleep is explicitly disabled by BT in case the laptop is charging. Once it disables charging, sleep is re-enabled.
3. Yes. Plugging-in also is event-based, so it will just idle, really.

closed 2024-10-22 by mhaeuser

Sangeppato, 2024-10-22:
Understood, thank you!