mhaeuser/Battery-Toolkit #47: pmset mode 25 cause battery toolkit not working

erictioh, 2024-11-05:
After i changed to mode 25, the battery toolkit will not work, it will allow the charge to go to 100% despite setting at 70%.

To set to mode 25 i did this in terminal.

sudo pmset hibernatemode 25

mhaeuser, 2024-11-23:
I have no idea whatsoever how hibernatemode would change anything. My best guess is you didn’t disable Optimized Charging.

mhaeuser, 2025-02-24:
Is there any update to this?

erictioh, 2025-02-25:
battery optimization was off (in fact both on / off i have tried and has changes to the issue).

it happens after going to hibernate then wakeup and battery will charge to 100% despite battery toolkit is set to 70.

i have switch to other AlDente and no such issue encountered.

mhaeuser, 2025-03-05:
Only reason I can think of is text when you hibernate while unplugged, then plug it in, and then wake it, XNU does not send a plugged-in event. I would consider this an XNU bug, but oh well. I’ll try to test this eventually.

mhaeuser, 2025-03-18:
I assume this to be fixed.

closed 2025-03-18 by mhaeuser