mhaeuser/Battery-Toolkit #54: Battery settings are stuck after uninstalling the app

boramerts, 2024-11-22:
I've installed the app yesterday and I've enabled settings as below. Today, I've uninstalled the app but now my M4 MacBook Pro is not charging above 76%

[image]

I've tried multiple plugs, my charge adapter and magsafe cable are original but it always "Battery is not charging"

[image]

mhaeuser, 2024-11-22:
I’m surprised I keep getting such reports, because it is literally impossible. Past Ventura, all executables remain within the app bundle and Apple Silicon SMC is stateless, so when you delete the app bundle and reboot, there is no technical possibility Battery Toolkit interferes with anything. Use Activity Monitor to locate the app bundle location.

closed 2024-11-22 by mhaeuser

N0ne1eft, 2025-01-13:
Check whether you have emptied Trash. Just tried to replicate this issue myself and I discovered that the daemon will still be running from Trash, emptying the Trash followed by a restart allows the charging to revert back to system control.
[image]