mhaeuser/Battery-Toolkit #55: "Sync Magsafe indicator light" no longer working?

ghost-bub, 2024-11-22:
I'm sure this is something I'm missing on my end, but I noticed a weird change with this feature that I would like to know what is different.

I recently updated my Macbook from an M3 Max to an M4 Max. Had this app on the M3 and currently on the M4. M3, the magsafe sync worked exactly as expected and documented.

Now? It doesn't sync charging status, so it isn't blinking anymore while charging. The only real change aside from the hardware has been It is running macOS 15.1.1 (just updated to this yesterday, was running 15.1 before and same issue).

Any insight at all would be helpful. Love this app.

mhaeuser, 2024-11-22:
It works just fine for me on M3 Pro, so I guess this indeed is a hardware change with M4. Not something I can really get into right now, as I lack the hardware (and time).

ghost-bub, 2024-11-22:
No worries. As far as I've seen, the only thing that isn't working is the sync indicator light. Just let me know if there's anything I can do on my end to help out.

loukhin, 2024-12-07:
I came across this issue while browsing and wanted to share my experience. On my Mac with the M4 Pro chip, I haven’t encountered this behavior. For example, the blinking light when charging and the amber light when holding work as expected (though I haven’t yet tested charging to full with MagSafe).

Could this behavior be tied to specific Mac models or configurations?

nolen8, 2024-12-24:
Mine stopped working on my 16" m3 pro as well! I did update to 15.2, however I don't know if that's the cause...

It blinks when first plugged in, however it immediately stops syncing and stays solid orange the whole time.

Charging and charge limit works as expected, just syncing isn't working correctly.

mhaeuser, 2025-03-05:
Quite possible macOS resets the status back to System at some events, but it's weird reports are inconsistent. I'll see when/if I can reproduce this and whether there will be more reports. Ir's not impossible other software is interfering.

karolpeszek, 2025-07-12:
Just want to confirm that It might be something with the hardware, since on macOS 15.5 and 16" MBP M4 Pro it also seems broken.

mhaeuser, 2025-08-08:
@tigris030 And it didn’t with Battery Toolkit? It’s weird, because in theory, the script does the exact same thing as Battery Toolkit.

The reports here vary wildly - works, doesn’t work at all, sometimes works and then stops, this script that superficially does the same thing works. I’m kind of lost, still works fine on my M3 Pro. :)

tigris030, 2025-08-08:
As we know:

* **Orange / Amber** → The battery is charging
* **Green** → The battery is fully charged (or not charging)

I’ve set my charge limit to 50%, with a lower limit of 40%. When the battery is at 48% and connected to the adapter on an M4 Pro, the indicator shows orange/amber.

Perhaps it’s impossible for it to show green while within that range? I know that “battery” only supports maintaining a fixed percentage and doesn’t allow lower limits. Could that be why it works with “battery” but not with your tool?

@mhaeuser

mhaeuser, 2025-08-08:
@tigris030 No, it will be green only if at 100 % (this actually matches what Apple does with Optimized Charging). Glowing amber is expected for being within the range and not charging. When charging, the light should blink.

tigris030, 2025-08-08:
> > [@tigris030](https://github.com/tigris030) No, it will be green only if at 100 % (this actually matches what Apple does with Optimized Charging). Glowing amber is expected for being within the range and not charging. When charging, the light should blink.
> 
> Thanks. It is intended behavior then. Yeah, not sure how he made it make glow green at 50 % (maintained value) then. This confused me.

This seems to be the same behavior for AIDente too.

> - Charge Limit reached or above it: Green
> - Charging paused because of Sailing Mode: Green

Do you think such feature could be implemented for your tool too @mhaeuser ?

tigris030, 2025-08-08:
> Yes, I could, but this actively loses information as you can no longer distinguish between partially charged and fully charged and I frankly don’t see the advantage.

Yeah, good points. Either way, if you want to include it as optional (even if it's hidden) option, I would use it. 😄