mhaeuser/Battery-Toolkit #58: What happens when "charge on below" and "charge off above" are equal?

uandere, 2024-11-29:
Just want to clarify. We have TWO options to modify:
[image]

I want to ask whether something 'bad' happens when "turn battery charging on below" and "turn battery charging off above" options are equal, say 80% (see screenshot). In this case, will the program's behaviour be "turning the battery off when it reaches exactly 80.00%, and then, just a few seconds after turning it on because the battery went down to 79.99%"? Or is it "charging up to 80% and then just hold it there providing just enough charge for that"?

Cause if it's the first option (the constant charge-on/charge-off loop), then I'm a bit concerned about it's "safety" for my battery. What I want is the second one (like on the new iPhones or some of Android phones), when I am able to just "freeze" the charging at 80%. 

Sorry if I sound stupid, but that's what I really need to clarify.

mhaeuser, 2024-11-29:
Charge levels are reported strictly in integers, so it would start charging again once it drops to 79%. Once charging is disabled, MacBooks keep running off AC, so the battery will not be utilized at all unless the AC provides too little energy to keep up. From there, the battery will still slowly discharge, but it will typically take several days (I think sometimes even weeks? Never thought about it much) to drop from 80% to 79%.

closed 2024-11-29 by mhaeuser

uandere, 2024-11-29:
Okay, so is it technically and physically safe for my battery to use your program that way?

mhaeuser, 2024-11-30:
Yes. It might not be considered best practice (as Apple themselves implemented a hysteresis for “trickle discharging”), but it should be fine.

uandere, 2024-12-04:
thank you man!