mhaeuser/Battery-Toolkit #66: MenuBar not opening? - build and run locally?

alexisgaziello, 2025-01-10:
Hey!

Thanks for the app. I was hoping to maybe contribute. Could you help me out building the project?

After cloning and running the project, the Dock icon shows up, but not the menu icon. Is there anything I should set up?

[image]

mhaeuser, 2025-01-11:
Probably welcome to the hellhole that is Apple codesign. In case you used the prebuilt app before, best run uninstall.sh and reboot before launching your own. Might be something is weirdly cached. Have you adjusted the Xcode project build variables to match your development certificate? I’m not at the Mac, so I can’t take screenshots right now, sorry: https://github.com/mhaeuser/Battery-Toolkit/blob/4b9a74bf1c31a57d78eb351b69fe09b861252f60/Battery%20Toolkit.xcodeproj/project.pbxproj#L875

alexisgaziello, 2025-01-11:
I did sign the app with my own certificate to be able to build the app, but I haven't modified the `BT_CODESIGN_CN` variable. What should I set it to? Why do we need this variable? Do I need to change anything else?

alexisgaziello, 2025-01-11:
I tried setting it to `Apple Development: Alexis Gaziello (244U9R97J4)` (replaced with my name and the number I got it from "DEVELOPMENT_TEAM", but its still not working.

mhaeuser, 2025-01-11:
Run ``codesign --display --verbose=4`` on your application bundle and use the ``Authority`` value (gives you the ``87DYA6FH9K`` from ``BT_CODESIGN_CN``). ``TeamIdentifier`` is entirely separate and is the ``EMH49F8A2Y`` prefix of ``BT_DAEMON_CONN``. I don't remember whether you actually need to change the latter, probably not unless you plan to notarize it. In case you do, it appears in various other locations as well, so grep for it.

This is required because daemon installation and communication is authenticated by the signature of the GUI app. I'm not aware of any Xcode features to automatically populate these values, so this awkward mess is what I am left with.

alexisgaziello, 2025-01-11:
Understood. 

Set up the value for the 5 targets, clean & built, but still no luck. Menu bar icon is not appearing.

mhaeuser, 2025-01-11:
Bizarre. Maybe check Console and make sure to wait a little, there are multiple retries to connect to the daemon. Did you get the prompt to install the daemon to begin with?