mhaeuser/Battery-Toolkit #75: SyncMagsafe indicator not working.

Psychodeus, 2025-02-16:
I came here after I updated my BatFi and all of sudden demand payment -.-
So yeah, the light sync isn't working for magsafe.
[image]

[image][image]

mhaeuser, 2025-02-16:
How is it not working? As stated, it's static amber when on hold, it is indeed on hold, and I see an amber light. Is it flashing?

Psychodeus, 2025-02-24:
@mhaeuser Not flashing

mhaeuser, 2025-02-24:
Well, then it works as described.

closed 2025-02-24 by mhaeuser