mhaeuser/Battery-Toolkit #80: "Apple could not verify the app is free of malware"

TPerformance, 2025-02-28:
[image]

Hi, I just wanted to ask if the above mentioned notification is the expected behavior. I'm sorry if I that's mentioned somewhere and I just didn't read it.

Kind regards

TPerformance, 2025-02-28:
I found another request where you mentioned that you don't have the Apple Developer Program Membership. So that concludes my request. Sorry for the trouble.
What's the cost for this membership? Would a few supporters be enough to cover for it?

Kind regards

Stoppels, 2025-03-02:
Workarounds for those who don't know:

On [macOS Sonoma 14](https://support.apple.com/guide/mac-help/mh40616/14.0/mac/14.0) and previous versions you can right-click/control-click the app and select Open. This adds the app as an exception.

On [macOS Sequoia (15)](https://support.apple.com/guide/mac-help/mh40616/mac) only the option in System Settings remains:

First try to open the app, macOS will gaslight you with the dialog above.

Go to **System Settings** → **Privacy & Security** → **Scroll** all the way down to Security → Click **Open** → Click **Open Anyway**

daalfa, 2025-03-03:
Additionally on Sequoia (15), the **Open Anyway** option will only be available after you click **Done** and close the app.

Dids, 2025-03-06:
Removing the quarantine flag manually tends to be faster than the alternatives:
```bash
sudo xattr -rd com.apple.quarantine /Applications/Battery\ Toolkit.app
```

realrogue, 2025-03-12:
> Removing the quarantine flag manually tends to be faster than the alternatives:
> 
> sudo xattr -rd com.apple.quarantine /Applications/Battery\ Toolkit.app
> One might argue that this circumvents and even disables security features, but as long as you trust the source, it should be fine.

I would add, [if installing via homebrew](https://github.com/mhaeuser/homebrew-mhaeuser), you can use the `--no-quarantine` flag.

mhaeuser, 2025-03-18:
The README now sufficiently clarifies the status quo and I don't plan to get a membership any time soon. I, in theory, have someone to notarize it for me, but that would require a convenient separation of development and production workflows - also not any time soon.

closed 2025-03-18 by mhaeuser