mhaeuser/Battery-Toolkit #83: Can't put Macbook on sleep - related?

MacTechTips, 2025-03-08:
Hi, first, I really love this utility, been using it for a few months on a Macbook Pro M4. Here's something I experience which may or may not be related to Battery Toolkit. A few times, my computer wasn't able to go to sleep mode, the option was greyed out. Looks like disabling Battery Toolkit from running in the background solved it. But may be a coincidence.

First times rebooted, didn't help, but removing battery toolkit entirely seemed to bring back the sleep function. I reinstalled Battery Toolkit after, worked fine. This happened again yesterday, but I didn't uninstall. Instead, I just went to "Login Items & Extensions", and in "Allow in the Background" toggled Battery Toolkit off. That would bring back the sleep function immediately. Turning if on would again make sleep unavailable. Could there be a relation? Or just a pure coincidence? A while later, seemed to work normally with Battery Toolkit on.

NOTE: This behavior seems to have happened after running resource-intensive activities on the Mac, like rendering AI images locally, using 100% of the GPUs, with fans running full speed. Maybe sleep is off until the temperature goes down? But why would turning off Battery Toolkit off in the background immediately bring the sleep function availability back?

mhaeuser, 2025-03-09:
Battery Toolkit disables sleep while it is charging, because it has to actively disable charging once reaching the maximum. I’m not aware of any other truly reliable way.

MacTechTips, 2025-03-10:
Ouch, I think that's not documented? So that means if not charged to the upper limit set in Battery Toolkit, someone can't put their computer to sleep (unless they go to login and extensions to turn it off)? Or does closing the lid still force the laptop to sleep? If not, it might be a problem. Let's take a use case, where the laptop isn't charged, and the user needs to leave (without their charger) and plan to use the laptop on battery later. The laptop won't be able to go into sleep mode, and their battery is draining even closed. And they'll never be able to put it to sleep again until they charge. Not many people know about Login items & Extensions. 

It might be a good idea to find a way to let users know what's happening, to manage their expectations, with upfront information so they know Battery Toolkit is preventing the computer from going into sleep mode. And maybe allow them to bypass that and put the computer to sleep anyway (I think that's essential). Not sure it's the best way, would need to think about it, but an option in the menu to "Allow sleep even when not charged" would serve both purposes; inform users that BT prevents sleep when not fully charged, and well as allow them to put their computer in sleep mode. 

Just a thought, don't take it personally. I can't be the only one who ran into that problem, unable to put the computer to sleep, and no idea why. First time it sent me into a lengthy search online, ended up changing pref files, entering some commands in terminal to try to reset settings and force sleep, only in the end to try and uninstalling Battery Toolkit entirely. It's such a useful tool, would be a shame people uninstall it permanently thinking it causes the computer to permanently lose sleep mode functionality.

mhaeuser, 2025-03-10:
No, I meant what I said precisely - sleep is disabled *while charging*. When you take it on the go, it is not charging, therefore sleep is not disabled. Anything else is a rare bug I never ran into (and frankly don’t know of anyone either and as you can see there currently is no similar issue filed) or a collision with another software that also disables sleep.

MacTechTips, 2025-03-10:
Still not obvious to everyone. Just saying. 🤷🏻‍♂️

mhaeuser, 2025-03-10:
Yes, I’m well aware, but I get the time to work on this further a couple of times a year, so I have to prioritise things. I took last week mostly off, so I could address a thing or two, but e.g. proper uninstall instructions have been on my TODO for like 2 years now, yet they keep getting deprioritised as I only get that question every couple of months.

Did I understand it right that sleep was never actually disabled while not charging and this was just a hypothesis?

MacTechTips, 2025-03-10:
I understand an uninstall might be a priority if it's been a long time waiting. I'll try to do more precise testing of the issue when I have a chance and report back the findings. Last time this happened (two days ago), the Macbook was plugged in, I haven't tried unplugging it to see if sleep become enabled.

mhaeuser, 2025-03-15:
README and the Settings GUI have been updated. I don't expect there to be sleep issues outside of charging, hence closing.

closed 2025-03-15 by mhaeuser

MacTechTips, 2025-03-26:
Sorry for the late reply, haven't had a chance to test. Thanks for the update, installed the new version. I really appreciate your app, and recommend it.