mhaeuser/Battery-Toolkit #94: battery still draining when running on power adapter with charging on hold

sourceofuncertainty, 2025-03-26:
battery has started to drain over time when running solely off power adapter with charging on hold. ive been using toolkit for several weeks and have only just now observed this behavior. i have verified that charging is indeed on hold and that the power source should be power adapter. running a 2021 MBP M1 Max on Sequoia 15.2. unplugging and plugging back in seems to fix the issue for now, but thought it might be worth it to share.

mhaeuser, 2025-03-26:
This is very much possible, especially with the Max during GPU workloads. The MBP may draw power from both the adapter and the battery to satisfy the current power requirement.

closed 2025-03-26 by mhaeuser

sourceofuncertainty, 2025-03-27:
makes sense, thanks for the reply. any idea why it might randomly charge a few percentage points past the set maximum seemingly randomly? my max is set to stop at above 79 (so 80 is max) and when i woke up it was charged to 85 (was not running any intensive process overnight.

mhaeuser, 2025-03-27:
I’ve sent his before but only *super* rarely. I can only guess the charge percentage is recalibrated some time. Make sure Optimized Charging is off and please report back if it happens frequently.