mhaeuser/Battery-Toolkit #96: Understanding Problem - help wanted

carepack, 2025-03-28:
Hi,

sorry, haven't found a better place for the question. Is there a better place?

First of all - thx for the app. Very useful and smart. With the latest release I can't get the idea behind the new values in settings. Threshold for start charging below can't be adjusted under 50%.
Why isn't possible to set the vale to 20%?

Thx for explaining.

mhaeuser, 2025-03-28:
There is the Discussions tab, but it doesn’t matter much. I don’t remember why I chose these values, but I changed it in the main branch to 20 and that will ship next update.

closed 2025-03-28 by mhaeuser

carepack, 2025-03-28:
thx @mhaeuser for the info. Ive been afraid of totally misunderstood the charging concept :-) and idea of this nice piece of software. Is there any possibility to buy you a coffee? This application is so useful.... Thanks a lot

mhaeuser, 2025-03-28:
Thanks for the kind words! No, I think you got it. :) The more I think about it, the more I feel like these values simply were a gut feeling. Like, if you set the minimum to 20 (Low Power Mode area) and attached it to power, it felt weird people want it to not charge at that point, so I simply took the minimum upper threshold - 5. But in any case, people wanted to lower it, so that will ship next update.

I don’t take “donations” because that really is taxed income in Germany. You can donate to this place, if you want to, but it’s SEPA-only (hence I didn’t bother linking it anywhere yet): https://www.kinderschutzbund-kaiserslautern.de/

carepack, 2025-03-29:
:-). Oh, yeah really? Isn't it best practice for battery life the 20 / 80 relation?! No matter why, 20 and 80 fits best for my case so thx that's available again soon.

I'll come over to kinderschutzbund with the donation. The topic matters in general and the battery toolkit spirit to forward the donation to them is very honorable.

thx for the friendly interaction and exchange :-) and with best regards

mhaeuser, 2025-03-29:
That’s a region you should not leave too often, not a region you should forcibly/continuously cycle within. 75/80 are exactly the Apple defaults for Optimized Charging as well.

Well, regarding the donation, I should point out my affiliation, so this is more of an ad. :) Thanks!

carepack, 2025-03-29:
Maybe lost in translation and I‘m not sure if i understood correct.

My config is simple. When the charger is plugged in its powering the macbook. If its unplugged I‘m running on battery. Charging starts on 20% and stops on 80%. In between start charging is a manual task and the power supply powers the MacBook  without charging the battery.

Charging cycles reduced to the min and the battery goes easy.

If I‘m wrong with this concept? Thx again

mhaeuser, 2025-03-29:
It doesn’t reduce cycles - whether you charge from 75 to 80 % 12 times or from 20 to 80 % once, both yield 0.6 cycles. The only point is to avoid trickle charging.