mhaeuser/Battery-Toolkit #100: Icon does not disappear from dock

Finn-Ko, 2025-04-01:
With the newest version (1.6) the Battery Toolkit icon does not disappear from the MacOS dock anymore after restarting my Mac. When opening the application settings and quitting the application (confirming that it will run in the background), the icon will shortly disappear and reappear a couple of seconds later.

mhaeuser, 2025-04-01:
Cannot reproduce. Please try the latest build with updated logging and monitor `Console` output for `me.mhaeuser.batterytoolkitd` entries: [Battery-Toolkit-40a2893.zip](https://github.com/user-attachments/files/19557183/Battery-Toolkit-40a2893.zip) \[ https://github.com/mhaeuser/Battery-Toolkit/commit/40a28931da73aeaa36cf884e053358d54b4dbdaf ]

Finn-Ko, 2025-04-01:
```

default	00:04:11.622497+0200	Battery Toolkit	order window: 237 op: 1 relative: 0 related: 0
default	00:04:11.623162+0200	Battery Toolkit	parent <NSStatusBarWindow: 0x121e28210> windowNumber=1e5 add <NSPopupMenuWindow: 0x11fe1b5f0> windowNumber=237 to group
default	00:04:15.546686+0200	Battery Toolkit	parent <NSStatusBarWindow: 0x121e28210> windowNumber=1e5 remove <NSPopupMenuWindow: 0x11fe1b5f0> windowNumber=237 from group
default	00:04:15.546914+0200	Battery Toolkit	window <NSPopupMenuWindow: 0x11fe1b5f0> windowNumber=237 finishing close
default	00:04:15.550777+0200	Battery Toolkit	order window: 237 op: 0 relative: 0 related: 0
default	00:04:15.605931+0200	Battery Toolkit	Window <NSWindow: 0x121911d10> windowNumber=238 ordered front from a non-active application and may order beneath the active application's windows.
default	00:04:15.605958+0200	Battery Toolkit	order window front conditionally: 238 related: 0
default	00:04:15.606342+0200	Battery Toolkit	establishing connection to agent
default	00:04:15.606386+0200	Battery Toolkit	[0x60000232fc50] Session created.
default	00:04:15.606405+0200	Battery Toolkit	[0x60000232fc50] Session created from connection [0x600003f303c0]
default	00:04:15.606425+0200	Battery Toolkit	[0x600003f303c0] activating connection: mach=true listener=false peer=false name=com.apple.uiintelligencesupport.agent
default	00:04:15.606534+0200	Battery Toolkit	[0x60000232fc50] Session activated
default	00:04:15.639554+0200	Battery Toolkit	SetFrontProcess: asn=0x0-0x6a06a options=0
default	00:04:15.645323+0200	Battery Toolkit	order window front conditionally: 238 related: 0
default	00:04:15.649631+0200	Battery Toolkit	order all windows front 5940356826500
default	00:04:15.650739+0200	Battery Toolkit	void _updateToReflectAutomaticTerminationState(void) Setting _kLSApplicationWouldBeTerminatedByTALKey=0
default	00:04:15.755620+0200	Battery Toolkit	order window front conditionally: 238 related: 0
default	00:04:18.123903+0200	Battery Toolkit	Service <private> status: 1
default	00:04:18.132855+0200	Battery Toolkit	Unregister (<private>) error: 1
default	00:04:18.132894+0200	Battery Toolkit	Failed to unregister login item: Error Domain=SMAppServiceErrorDomain Code=1 UserInfo={NSLocalizedFailureReason=<private>}
default	00:04:18.138989+0200	Battery Toolkit	Power settings have not changed, ignoring
default	00:04:18.150618+0200	Battery Toolkit	order window front conditionally: 238 related: 0
default	00:04:18.150738+0200	Battery Toolkit	parent <NSWindow: 0x121911d10> windowNumber=238 remove <_NSAlertPanel: 0x121b57960> windowNumber=239 from group
default	00:04:18.150764+0200	Battery Toolkit	parent <NSWindow: 0x121911d10> windowNumber=238 add <_NSAlertPanel: 0x121b57960> windowNumber=239 to group
default	00:04:18.151638+0200	Battery Toolkit	order window: 239 op: 1 relative: 238 related: 0
default	00:04:18.414791+0200	Battery Toolkit	parent <NSWindow: 0x121911d10> windowNumber=238 remove <_NSAlertPanel: 0x121b57960> windowNumber=239 from group
default	00:04:18.415035+0200	Battery Toolkit	parent <NSWindow: 0x121911d10> windowNumber=238 add <_NSAlertPanel: 0x121b57960> windowNumber=239 to group
default	00:04:18.415452+0200	Battery Toolkit	order window: 239 op: 1 relative: 238 related: 0
default	00:04:19.227250+0200	Battery Toolkit	order window front conditionally: 238 related: 0
default	00:04:19.489659+0200	Battery Toolkit	parent <NSWindow: 0x121911d10> windowNumber=238 remove <_NSAlertPanel: 0x121b57960> windowNumber=239 from group
default	00:04:19.491877+0200	Battery Toolkit	order window: 239 op: 0 relative: 0 related: 0
default	00:04:19.495410+0200	Battery Toolkit	order window: 239 op: 0 relative: 0 related: 0

```

Finn-Ko, 2025-04-01:
Especially relevant is probably
```

default	00:04:18.123903+0200	Battery Toolkit	Service <private> status: 1
default	00:04:18.132855+0200	Battery Toolkit	Unregister (<private>) error: 1
default	00:04:18.132894+0200	Battery Toolkit	Failed to unregister login item: Error Domain=SMAppServiceErrorDomain Code=1 UserInfo={NSLocalizedFailureReason=<private>}
default	00:04:18.138989+0200	Battery Toolkit	Power settings have not changed, ignoring

```

mhaeuser, 2025-04-01:
Sorry, the log is censored, please try again: [Battery-Toolkit-661c3d0.zip](https://github.com/user-attachments/files/19558772/Battery-Toolkit-661c3d0.zip) \[ https://github.com/mhaeuser/Battery-Toolkit/commit/661c3d0a4917526ed779550f6b57943409c37f29 ]

Finn-Ko, 2025-04-02:
```
default	20:29:15.037327+0200	Battery Toolkit	NotifyToken::RegisterDispatch(user.uid.501.com.apple.LaunchServices.database) fired for session key <private>
default	20:29:15.062685+0200	Battery Toolkit	Service <private> status: 1
default	20:29:15.070472+0200	Battery Toolkit	Unregister (<private>) error: 1
default	20:29:15.070502+0200	Battery Toolkit	Failed to unregister login item: Error Domain=SMAppServiceErrorDomain Code=1 "Operation not permitted" UserInfo={NSLocalizedFailureReason=Operation not permitted})
default	20:29:15.075170+0200	Battery Toolkit	[0x155ab0960] activating connection: mach=true listener=false peer=false name=com.apple.iconservices
default	20:29:15.075519+0200	Battery Toolkit	[0x144613180] activating connection: mach=true listener=false peer=false name=com.apple.iconservices.store
default	20:29:15.088236+0200	Battery Toolkit	Power settings have not changed, ignoring
default	20:29:15.099827+0200	Battery Toolkit	order window front conditionally: 805 related: 0
default	20:29:15.100045+0200	Battery Toolkit	parent <NSWindow: 0x1558101a0> windowNumber=805 remove <_NSAlertPanel: 0x155ab3ba0> windowNumber=809 from group
default	20:29:15.100069+0200	Battery Toolkit	parent <NSWindow: 0x1558101a0> windowNumber=805 add <_NSAlertPanel: 0x155ab3ba0> windowNumber=809 to group
default	20:29:15.101147+0200	Battery Toolkit	order window: 809 op: 1 relative: 805 related: 0
default	20:29:15.363235+0200	Battery Toolkit	parent <NSWindow: 0x1558101a0> windowNumber=805 remove <_NSAlertPanel: 0x155ab3ba0> windowNumber=809 from group
default	20:29:15.363373+0200	Battery Toolkit	parent <NSWindow: 0x1558101a0> windowNumber=805 add <_NSAlertPanel: 0x155ab3ba0> windowNumber=809 to group
default	20:29:15.363636+0200	Battery Toolkit	order window: 809 op: 1 relative: 805 related: 0
default	20:29:15.368868+0200	Battery Toolkit	[0x155904cf0] activating connection: mach=true listener=false peer=false name=mul-xpc (Apple)_OpenStep
default	20:29:15.369106+0200	Battery Toolkit	[0x1547078f0] activating connection: mach=true listener=false peer=false name=com.apple.naturallanguaged
default	20:29:16.822295+0200	Battery Toolkit	order window front conditionally: 805 related: 0
default	20:29:17.084691+0200	Battery Toolkit	parent <NSWindow: 0x1558101a0> windowNumber=805 remove <_NSAlertPanel: 0x155ab3ba0> windowNumber=809 from group
default	20:29:17.086289+0200	Battery Toolkit	order window: 809 op: 0 relative: 0 related: 0
default	20:29:17.090357+0200	Battery Toolkit	order window: 809 op: 0 relative: 0 related: 0

```

mhaeuser, 2025-04-02:
Any chance this is some aftermath of the Developer ID change? https://github.com/mhaeuser/Battery-Toolkit/releases/tag/1.3
In that case, running [uninstall.sh](https://github.com/mhaeuser/Battery-Toolkit/blob/main/uninstall.sh), deleting the app, rebooting, and reinstalling should do the trick.

Finn-Ko, 2025-04-02:
The error persists

mhaeuser, 2025-04-02:
Average Apple cache db corruption bug...

closed 2025-04-02 by mhaeuser