mhaeuser/Battery-Toolkit #98: does not stop charging for me

hexeric, 2025-04-01:
...only if i press 'disable power adapter' in the app. whenever the battery is under 80% it charges, ignores the settings and only stops for a little while at 80, continues charging after a few minutes. just had a fresh install of BT and it did work before. probably something broke after the last macOS updates? i am still on Sonoma! MBP M1 pro 2021 with macOS 14.7.5

mhaeuser, 2025-04-01:
15.4 works fine and I have not gotten any similar reports yet. We’ll see…

hexeric, 2025-04-01:
i'm on sonoma, not sequoia

mhaeuser, 2025-04-03:
I confirmed with multiple people that they hit no such bug. Are you sure Optimized Charging is disabled and no other app messes with charging?

hexeric, 2025-04-04:
unfortunately yes. it worked well for weeks, stopped doing so after the latest OS update. optimized charging is off and no other battery app was installed. charging comes and goes but does not stop completely. it works as intended when using one of the other apps out there

Rick02022, 2025-05-11:
I'm also on Sonoma 14.7.5 (Macbook Air M2), and it works fine for me.

mhaeuser, 2025-05-19:
Closing due to lack of reports.

closed 2025-05-19 by mhaeuser

john20xdoe, 2025-06-20:
@mhaeuser it is happening again right now. 
Here's the Battery Toolkit UI:

[image]

Here In the macos native battery UI, it shows Not Charging. But here in the screenshot, it has creeped up past 80% again:

[image]

My Battery Toolkit Settings:

[image]