mhaeuser/Battery-Toolkit #99: Can't access Mac with external kb/m from lock screen while discharge is active

Alexrocks1253, 2025-04-01:
I believe that the prevent sleep option may be broken. I use a docked setup, and I can't get the computer to the lock screen with the keyboard or mouse while the app is active. What I think may be happening is that the computer is sleeping or that something is preventing the mouse and keyboard from connecting. The lock button on the keyboard seems to force sleep a few seconds after the lock screen shows.

mhaeuser, 2025-04-01:
Cannot reproduce - can you open the lid and see whether "Sleep" is grayed out in the Apple menu? Other software might interfere and re-enable sleep.

Alexrocks1253, 2025-04-02:
Sleep is grayed out when the lid is open or closed while discharging is on.

[image]

Alexrocks1253, 2025-04-02:
So my computer is in a sort of dock configuration where the main power source is the USB-C coming from an ASUS ProArt monitor. The monitor provides enough current to charge the laptop so I don't use the Magsafe charger.

For me to reproduce:
1. Enable discharge
2. Lock using the keyboard's touch ID button
3. Magic Keyboard and Logitech MX Master 3S can't get the computer on unless I open the lid, then close the lid again.

mhaeuser, 2025-04-02:
Right, now I can reproduce it. Feels like an Apple bug. I’ll try a thing or two when I get the time, but it’s best to simply not lock it like that for the time being.

Alexrocks1253, 2025-04-03:
Ok good to know, thank you.

antoniovalentino, 2025-04-16:
I have the same issue as @Alexrocks1253, sleep grayed out and Logitech MX Master 3S can't get the computer on unless I open the lid