mhaeuser/Battery-Toolkit #105: Could not find and/or execute program specified by service: 128: The specified path is not a bundle: Contents/Library/LaunchServices/me.mhaeuser.batterytoolkitd

ivankovnatsky, 2025-04-11:
👋 @mhaeuser, thanks for taking the time working on this application!

Recently I started to have issues on one of my laptops:

```console
log show --predicate 'subsystem contains "me.mhaeuser"' --last 1m
...
2025-04-11 16:34:36.631235+0300 0x7f5ad1   Default     0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd [64885]:] Could not find and/or execute program specified by service: 128: The specified path is not a bundle: Contents/Library/LaunchServices/me.mhaeuser.batterytoolkitd
2025-04-11 16:34:36.631261+0300 0x7f5ad1   Error       0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd [64885]:] Service could not initialize: copy_bundle_path(E9210303-5274-42F6-A916-1398173ABEAB, 0, 0), error 0x6f - Invalid or missing Program/ProgramArguments
2025-04-11 16:34:36.631267+0300 0x7f5ad1   Error       0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd [64885]:] initialization failure: 24E248: xpcproxy + 23288 [475][86AEEF9E-8EB2-3E8C-AE63-66AC6C5F572B]: 0x6f
2025-04-11 16:34:36.631272+0300 0x7f5ad1   Default     0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd [64885]:] internal event: INIT, code = 111
2025-04-11 16:34:36.631276+0300 0x7f5ad1   Default     0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd [64885]:] job state = spawn failed
2025-04-11 16:34:36.632658+0300 0x7f5ad1   Default     0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd [64885]:] xpcproxy exited due to exit(78)
2025-04-11 16:34:36.632664+0300 0x7f5ad1   Default     0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd [64885]:] exited due to exit(78), ran for 55ms
2025-04-11 16:34:36.632668+0300 0x7f5ad1   Default     0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd [64885]:] already handled failed init, ignoring
2025-04-11 16:34:36.632673+0300 0x7f5ad1   Default     0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd [64885]:] service state: exited
2025-04-11 16:34:36.632680+0300 0x7f5ad1   Default     0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd [64885]:] internal event: EXITED, code = 0
2025-04-11 16:34:36.632705+0300 0x7f5ad1   Default     0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd [64885]:] service state: not running
2025-04-11 16:34:36.632711+0300 0x7f5ad1   Default     0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd:] Service only ran for 0 seconds. Pushing respawn out by 10 seconds.
2025-04-11 16:34:36.632813+0300 0x7f5ad1   Default     0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd:] internal event: WILL_SPAWN, code = 0
2025-04-11 16:34:36.632818+0300 0x7f5ad1   Default     0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd:] service state: spawn scheduled
2025-04-11 16:34:36.632821+0300 0x7f5ad1   Default     0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd:] service spawn deferred by 10 seconds due to throttle
2025-04-11 16:34:46.634864+0300 0x7f5dc6   Default     0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd:] service state: spawning
2025-04-11 16:34:46.634892+0300 0x7f5dc6   Default     0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd:] launching: semaphore
2025-04-11 16:34:46.634906+0300 0x7f5dc6   Default     0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd:] Allowing non-reentrant proxy for resolving path
2025-04-11 16:34:46.637212+0300 0x7f5dc6   Default     0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd [64894]:] xpcproxy spawned with pid 64894
2025-04-11 16:34:46.637249+0300 0x7f5dc6   Default     0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd [64894]:] internal event: SPAWNED, code = 0
2025-04-11 16:34:46.637254+0300 0x7f5dc6   Default     0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd [64894]:] service state: xpcproxy
2025-04-11 16:34:46.637259+0300 0x7f5dc6   Default     0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd [64894]:] deferred event: domain spawn response: 0
2025-04-11 16:34:46.637435+0300 0x7f5dc6   Default     0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd [64894]:] internal event: SOURCE_ATTACH, code = 0
2025-04-11 16:34:46.684138+0300 0x7f5c87   Default     0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd [64894]:] Could not find and/or execute program specified by service: 128: The specified path is not a bundle: Contents/Library/LaunchServices/me.mhaeuser.batterytoolkitd
2025-04-11 16:34:46.684155+0300 0x7f5c87   Error       0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd [64894]:] Service could not initialize: copy_bundle_path(E9210303-5274-42F6-A916-1398173ABEAB, 0, 0), error 0x6f - Invalid or missing Program/ProgramArguments
2025-04-11 16:34:46.684159+0300 0x7f5c87   Error       0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd [64894]:] initialization failure: 24E248: xpcproxy + 23288 [475][86AEEF9E-8EB2-3E8C-AE63-66AC6C5F572B]: 0x6f
2025-04-11 16:34:46.684163+0300 0x7f5c87   Default     0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd [64894]:] internal event: INIT, code = 111
2025-04-11 16:34:46.684167+0300 0x7f5c87   Default     0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd [64894]:] job state = spawn failed
2025-04-11 16:34:46.685714+0300 0x7f5c87   Default     0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd [64894]:] xpcproxy exited due to exit(78)
2025-04-11 16:34:46.685720+0300 0x7f5c87   Default     0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd [64894]:] exited due to exit(78), ran for 49ms
2025-04-11 16:34:46.685723+0300 0x7f5c87   Default     0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd [64894]:] already handled failed init, ignoring
2025-04-11 16:34:46.685727+0300 0x7f5c87   Default     0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd [64894]:] service state: exited
2025-04-11 16:34:46.685733+0300 0x7f5c87   Default     0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd [64894]:] internal event: EXITED, code = 0
2025-04-11 16:34:46.685755+0300 0x7f5c87   Default     0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd [64894]:] service state: not running
2025-04-11 16:34:46.685760+0300 0x7f5c87   Default     0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd:] Service only ran for 0 seconds. Pushing respawn out by 10 seconds.
2025-04-11 16:34:46.685984+0300 0x7f5c87   Default     0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd:] internal event: WILL_SPAWN, code = 0
2025-04-11 16:34:46.685988+0300 0x7f5c87   Default     0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd:] service state: spawn scheduled
2025-04-11 16:34:46.685990+0300 0x7f5c87   Default     0x0                  1      0    launchd: [system/me.mhaeuser.batterytoolkitd:] service spawn deferred by 10 seconds due to throttle
--------------------------------------------------------------------------------------------------------------------
Log      - Default:        126, Info:                0, Debug:             0, Error:         12, Fault:          0
Activity - Create:           0, Transition:          0, Actions:           0
~ 
at Apr 11 16:35 ❯ 
```

This issue on M1 Pro, M2 Air and M3 Pro are running it fine.

Normally I package the app with nix overlay, but to reproduce the issue I tried downloading it manually, running from Downloads directory and also placing the app under /Applications, same thing.

OS: 15.4.

I previously run the app on this laptop pretty much fine, I don't remember why but I think I remove some time before and installed again a bit later.

Do you know maybe what might be the issue?

mhaeuser, 2025-04-11:
Hi, sorry, never seen this before. Sometimes, macOS has borked caches like BTM that cause issues with reinstalling and such. See this comment on how to deep-uninstall and then try anew: https://github.com/mhaeuser/Battery-Toolkit/issues/7#issuecomment-2036664963

ivankovnatsky, 2025-04-11:
In my case I had to play a bit more, garbage collecting nix store, but yeah, it worked, thanks!

closed 2025-04-11 by ivankovnatsky

mhaeuser, 2025-04-11:
I have no Earthly idea how everything is so fragile and how BTM cannot be wiped with a simple command, but oh well… glad it worked.