mhaeuser/Battery-Toolkit #107: [Feature] Hide Menubar Icon

musamatini, 2025-04-22:
Hello,

I usually use the app to set a battery charge limit and then forget about it. It would be great if there were an option in the settings to hide the icon from the menu bar. To bring it back, the user could reopen the app, which would open the settings so they can easily turn the icon back on. Currently, I’m using Hidden Bar to hide the icon, but it still leaves an extra arrow to show hidden icons, which is kind of annoying.

Thanks for your great work—Amazing app!

dhinakg, 2025-04-22:
https://github.com/mhaeuser/Battery-Toolkit/issues/106#issuecomment-2812107335

iamsoum, 2025-04-22:
Use ICE which is basically free alternative to bartender. You can use it to hide any app's icon from menu bar.

mhaeuser, 2025-04-22:
See @dhinakg’s comment.

closed 2025-04-22 by mhaeuser

musamatini, 2025-04-22:
Should have checked the closed issues before adding a comment, i am really sorry.