mhaeuser/Battery-Toolkit #108: The commands are not working as expected.

iamsoum, 2025-04-22:
Mine's at 80% now. I want to charge it to full. So I click on the icon in the menu bar and click "Charge to full now". Didn't do anything so I clicked "disable power adapter" and then "request charging to full". Power adapter is now disabled but the battery is still not charging. It's depleting I'm typing this and the same button of "charge to full" is now showing "request charging to 80% now". Why is it so complicated?

mhaeuser, 2025-04-22:
“Disable Power Adapter” behaves exactly as you should expect - no power adapter, no charging, so I don’t understand that part. Just leave it on.

“Charge to Full” should immediately start charging to 100 % (notice that there may be a many-second delay between starting to charge and macOS actually refreshing the charging status) and this should be indicated by a “Charging to Full” label at the top of the Battery Toolkit menu.

iamsoum, 2025-04-22:
I realized to leave the power adapter on. But I waited 10-20 seconds and I tried again but it didn’t work. I got it to charge to full by increasing the slider "turn battery charging off after" to 100.

mhaeuser, 2025-04-22:
It might very well take 30 seconds for the state to refresh sometimes - a little patience…