mhaeuser/Battery-Toolkit #115: wrong charging status in stand-by

rejetto, 2025-05-31:
thanks for your project!
i have my thresholds set to 75 and 80

[image]

i just switched on my mac, and i've found it at 100%.
i suspect it wentstandby in a moment where the battery was actively charging. 
do you think the same?
if that's the case, it would be cool to have (as an option?) the program to detect going standby, and IF within the thresholds, to disable charging.

mhaeuser, 2025-05-31:
Sleep is disabled while charging. This may happen if you didn’t disable Optimized Charging, enter hibernation, or shut down.

rejetto, 2025-05-31:
optimized charging was disabled

[image]

what i did, is to execute `iokit_power_management_sleep_system` with karabiner, and i assume it's equivalent to the sleep in the "mac" menu.

[image]

does this make sense to you?

mhaeuser, 2025-05-31:
You forced sleep while sleep was disabled by the tool - just don’t do that.