mhaeuser/Battery-Toolkit #116: "Failed to communicate with the background service" when building from source

bryango, 2025-06-02:
Thank you for this wonderful app! I would like to build and run it from source. The build command that I used is:
```bash
xcodebuild -scheme "Battery Toolkit" -configuration Release \
	CODE_SIGN_IDENTITY="Apple Development" \
	DEVELOPMENT_TEAM=AWMJ8H4G7B # my own dev id
```
and it seems to build fine. However, on start up it errors out with "Failed to communicate with the background service":
[image]

What did I do wrong? Are there any extra steps I should take, or is my build command simply incorrect?

mhaeuser, 2025-06-02:
That is definitely not enough. Building with a different certificate is poorly tested (and poorly supported by Apple), but some hints (without success, as the person vanished) can be found here: https://github.com/mhaeuser/Battery-Toolkit/issues/66

bryango, 2025-06-03:
Closing in favor of #79.

closed 2025-06-03 by bryango