mhaeuser/Battery-Toolkit #117: Error - Falsely incompatible mac

Kcraft059, 2025-06-09:
[image]

After having updated to macOS 26 beta on a separated volume (dual boot) and getting back to my volume running 15.4.1 the app says it's not compatible… any way to bypass it ? (Might be because of a firmware update sadly…)

Kcraft059, 2025-06-09:
Absolutely no problem, just asking, were my assumptions right about it being linked to firmware ? 
The thing that surprises me the most is that on macOS 15.5 even after updating my 2nd volume to macOS 26 the system is still able to control battery charging while the app isn't… maybe they're using different frameworks to control the hardware…

(And anyway, the dev beta 1 is quite bad so far it's really buggy and breaks a lot of things (like homebrew 😅 - tho nix to my surprise has successfully survived)

Sry for bothering you abt that btw…

mhaeuser, 2025-06-09:
Well, it's either SMC firmware or the SMC KEXT, hard to tell from just that error. If it still works with 15.5, it might actually just be the KEXT. Or somehow the OS volume dictates the firmware protocol version (the SoC basically just implements the protocol at this point, there is no dedicated SMC). Time will tell. :)

tnix100, 2025-06-10:
Same issue here with a MacBook Air M4. I installed  macOS 26 on an external SSD, now Battery Toolkit will not load on macOS 15 on my internal SSD, or macOS 26 on my external SSD.

mhaeuser, 2025-06-16:
The tool may be permanently dead: https://github.com/charlie0129/batt/issues/34#issuecomment-2975821260

But let’s wait and see…

js4jiang5, 2025-06-17:
> Another charge limiter app (aldente) supports macos 26 now: https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/releases/tag/1.33.2

Ths SMC key CH0B, CH0C for charging control is missing in macOS 26 Tahoe, and CH0I, CH0K for discharge is missing, too. I'm wondering how Aldente 1.33.2 fixed it.

ichbinshhan, 2025-06-20:
The situation still exists after MacOS 26 was downgraded to MacOS 15.

mhaeuser, 2025-08-06:
For now, fixed at https://github.com/mhaeuser/Battery-Toolkit/commit/64c97cc34896c95f2ee26f9ba5bed0e67ca6c5a3
If someone wants to help testing, the fix applies to both macOS 15.7 and 26:  [Battery-Toolkit-1.8-beta.zip](https://github.com/user-attachments/files/21618156/Battery-Toolkit-1.8-beta.zip)

ichbinshhan, 2025-08-06:
> For now, fixed at [64c97cc](https://github.com/mhaeuser/Battery-Toolkit/commit/64c97cc34896c95f2ee26f9ba5bed0e67ca6c5a3) If someone wants to help testing, the fix applies to both macOS 15.7 and 26: [Battery-Toolkit-1.8-beta.zip](https://github.com/user-attachments/files/21618156/Battery-Toolkit-1.8-beta.zip)

15.7????

mhaeuser, 2025-08-06:
@Kcraft059 Just report them here, thanks!

th0u, 2025-08-09:
> For now, fixed at [64c97cc](https://github.com/mhaeuser/Battery-Toolkit/commit/64c97cc34896c95f2ee26f9ba5bed0e67ca6c5a3) If someone wants to help testing, the fix applies to both macOS 15.7 and 26: [...]

Thank you! Works with MBAir M3 macos26 latest Dev Beta.

mhaeuser, 2025-08-11:
Released v1.8 at https://github.com/mhaeuser/Battery-Toolkit/releases/tag/1.8
Closing for the time being.