mhaeuser/Battery-Toolkit #120: My Mac is suddenly not supported, even though I am using a supported Mac.

LucasCountyEAS, 2025-06-12:
I've been using Battery Toolkit since I got this MacBook in December. My battery is indeed still at 100% capacity with an 84 cycle count. 

All of a sudden, today, Battery Toolkit decided it would be funny to "unsupport" my Mac. 

I reinstalled it, and it's still showing me this message. Any help is appreciated.

[image]

ShadozRose, 2025-06-12:
I wonder if you installed macos 26 and reinstalled macos 15,if so,that´s why you cant use it anymore.you need to do  a full DFU restore

Sachindagar8813, 2025-06-15:
> I did install MacOS 26, but I partitioned my SSD for 1 part MacOS 15, and the other part for MacOS 26. Just to see what was going on with the new update. I then deleted it, and the MacOS 15 installation wasn't touched.

same problem , when will it get fixed ?

mhaeuser, 2025-06-15:
Again, as I said in one of the other three reports, I’m not installing a Dev Beta 1 on my production machine. DFU and use a stable OS.

mhaeuser, 2025-06-15:
@Sachindagar8813 Once your Pull Request is ready.