mhaeuser/Battery-Toolkit #130: MacBook were very hot as power adapter connected

EJeer, 2025-07-21:
Today I faced strange issue, than battery was changed by the % it is set to changer max (70%), but my MacBook was very hot untill I unplug power adapter (same with USB-c port). Restart doesnt helped. The MacBook shows everything as it should - Power Source power adapter, Battery is not charging, etc. Everything was correct, but the MacBook was very hot. It seems that MacBook continues to receive full charging power to the battery, but it seems that it was discharged at the same time.

What helped that is after disconeting and doing restart - battery level showed 71%, not 70% as it was before disconnecting and restarting. So I thought maybe maybe there is some bug, that battery is already 71%, but it tries to hold it on 70%. So I changed limit to 72%. As I changed whole overheating is gone and issue is fixed and at the moment not coming back.

erhelvaci, 2025-08-07:
Same issue. Just was charging 10% while off, laptop was burning hot, what is going on?

tigris030, 2025-08-08:
> [@tigris030](https://github.com/tigris030) What kind of attention do you expect? This is an Apple bug.

If it is related to Apple then there is nothing to do.

ulrichloose, 2025-08-21:
I have no problems with "heat", but a small oddity (apple m3, 15.6.1 (24G90):
At 9 p.m. the state of charge was at 80% and it was no longer charged. Notebook closed and at night it was no longer used, but still connected to the mains. In the morning notebook unfolded and the charge level was at 82%. Optimized charging is turned off.
In the system settings, the usage history was displayed like this:

[image]

At some point around 2 a.m. Another 2% was charged. Why?

mhaeuser, 2025-08-21:
@ulrichloose Well, I can never tell for sure, but this is likely just some sort of re-calibration done on the fly or something. I’m quite sure I’ve seen it with (only) Optimized Charging before too.