mhaeuser/Battery-Toolkit #132: Feature Request: Support "Hidden Login Item" for Cleaner macOS Experience

Ross9295, 2025-07-28:
Hi!

Love the app, it is truly helpful!

I have a tiny feature request if that's ok - support of the Hidden Login Item.

Currently, when added to Login Items (System Settings → General → Login Items), the app appears in the Dock at login every time after I restart my mac.
I thought the Hidden Login would help, but the icon still appears in the Dock after restart.

I'm not a programmer, but I assume it is just some simple tweak that could be done in 5 minutes but would make a difference for me.

Thanks for building such a great tool!

Ross

crystall1nedev, 2025-08-31:
The behavior you are experiencing might be remnants of the app in the Dock after it launches. The following video should help clarify things:

https://github.com/user-attachments/assets/5dc3b6b1-9efd-4bfa-8790-3414a178c44f

- We start with Battery Toolkit closed (running the background helper still, but the menu bar icon is currently disabled.
- We then open Battery Toolkit. I haven't yet switched to a different app, so currently Battery Toolkit is providing the options for the menu bar.
- Finally, I click into the Safari window in the background. This causes Battery Toolkit to lose focus and the Dock icon to no longer show it as running.

What's left is the Battery Toolkit icon in the "Recents" section of the Dock.

mhaeuser, 2025-08-31:
Then I’m puzzled, I can’t manage to reproduce this. Even though it’s not quite the same point, I’ll re-open this issue to track the issue. But I can’t really work on this now anyhow.

nekename, 2025-09-01:
Here you go. As you can see, the icon disappears for a few seconds when I open the Battery Toolkit settings and exit them, but reappears almost immediately.

https://github.com/user-attachments/assets/e3b8b065-1f9e-4649-a122-7162917c8472

crystall1nedev, 2025-09-01:
What version of macOS are you on currently?

nekename, 2025-09-01:
15.6(.2 or something)