mhaeuser/Battery-Toolkit #135: Custom MagSafe indicator light

gandandev, 2025-08-14:
It would be great if I could customize the MagSafe indicator light color, such as green when fully charged to the maximum charge limit or amber during charging.

gandandev, 2025-08-21:
The main purpose of this app is to enforce a charging limit. For example, if the limit is set to 80%, the battery should be considered "fully charged" once it reaches that threshold.

I think it would be clearer if the charging status was shown as:
* Orange while charging toward the limit
* Green once the limit is reached and the battery is being held

This way, the colors consistently reflect the charging state within the set limit.

mhaeuser, 2025-08-21:
@gandandev You’re not the first to suggest this, I’m aware of this opinion. But I will not change the default to be objectively less expressive. Whether there will be a toggle some time - maybe, but I can’t dedicate much time to the tool right now.

gandandev, 2025-08-21:
Got it, thanks for clarifying. A toggle in the future would be perfect, but I get that priorities and time are limited.