mhaeuser/Battery-Toolkit #152: Missing support for MacOS Tahoe

anderslatif, 2025-09-18:
I updated to Tahoe which was released on September 15, 2025 and Battery-Toolkit doesn't work anymore.

After the update, it would run as normal and the upper left context menus were available but the icon in the top right did not show up.

I then manually updated to the newest release and was met with the following message.

[image]

I have also tried running the `uninstall.sh` script and reran it to the same effect.

I really enjoy this software and am not in a rush to have it fixed but wanted to create an issue for others to find.

---

Label: [Bug]

mhaeuser, 2025-09-18:
Well, you say “latest”, but which version exactly? Someone else had an outdated brew cache the other day and was stuck on an older release. Please be specific and actually make sure it’s that exact version.

anderslatif, 2025-09-19:
This is the latest version:

https://github.com/mhaeuser/Battery-Toolkit/releases/tag/1.8

It can't be an outdated brew cache since I didn't use homebrew to install it.

closed 2025-09-19 by anderslatif